CuriousScientist0/ADS1256 #1: issue with olimex esp32 poe and solution

mafiou86, 2023-12-11:
Dears,
The module don't compile for olimex esp32 card due to an error of initialization values. 
The following modification in "ADS1256.cpp", line 44, solve the issue:
before:
ADS1256::ADS1256(const byte DRDY_pin, const byte RESET_pin, const byte SYNC_pin, const byte CS_pin, float VREF) 
after:
ADS1256::ADS1256(const byte DRDY_pin_, const byte RESET_pin_, const byte SYNC_pin_, const byte CS_pin_, float VREF) :
DRDY_pin(DRDY_pin_), RESET_pin(RESET_pin_), SYNC_pin(SYNC_pin_), CS_pin(CS_pin_)
best regards

CuriousScientist0, 2023-12-15:
Thanks for your feedback! I will see if I can replace this code with your suggested line. I don't know if it would break the code for the 8 other microcontrollers that I already tested (with pass) the code with. 

It might sound silly, but could you explain the difference between the two lines? I have never met the syntax you suggested, so I would like to learn more about it. Thanks!

CuriousScientist0, 2024-02-08:
This issue seems to be solved thanks to a kind contributor. Please check it! 

https://github.com/CuriousScientist0/ADS1256/pull/7

closed 2024-02-08 by CuriousScientist0