CuriousScientist0/ADS1256 #3: Read register with R command not working in esp32 (Always showing 81)

srisankaran, 2024-01-29:


CuriousScientist0, 2024-01-29:
You did not state which ESP board... Also, do you change the register before reading it? If you don't write something to the register, it will always return the same value. Also, check your wiring and circuitry. I remember your nickname from YouTube and based on your question there, I am not 100% sure that you do everything correctly (no offence). 
I have an ESP32-WROOM-32 in front of me as I am writing this reply, and I can read and write _any_ registers and it works perfectly.

closed 2024-01-29 by CuriousScientist0

srisankaran, 2024-01-29:
I haven't changed the value of the register, so shouldn't it give the current default register value?

CuriousScientist0, 2024-01-30:
I am still convinced that there's something wrong on your side. I have the circuit in front of me and it works well. Incorrect wiring, using the incorrect SPI (especially on an ESP32) and so on... Since you haven't provided any details, no images, no schematics, I won't be able to help.