CuriousScientist0/ADS1256 #5: Adaptive PGA Control for Improved Small-Scale Voltage Measurement Precision

srisankaran, 2024-01-30:
It would be cool if there was a function to implement a dynamic function to monitor voltage levels and automatically adjust the Programmable Gain Amplifier (PGA) when readings fall below specified thresholds, such as 2.5V, 1.25V,... This intelligent control mechanism optimizes precision by fine-tuning the PGA settings, ensuring more accurate and reliable small-scale voltage measurements.

CuriousScientist0, 2024-01-30:
Go ahead and implement it. The code is open source.

closed 2024-01-30 by CuriousScientist0

srisankaran, 2024-01-30:
Why isn't it a good idea?

CuriousScientist0, 2024-01-30:
Do it, if you want, I won't do it for you.

srisankaran, 2024-01-30:
`/*________________________________________________________________________________________________________
ESP32 micro-Voltmeter using ADS1256 
________________________________________________________________________________________________________*/

/*INFO

  PIN CONFIG (ESP32)
    SPI *default pins:
    IN    - 23 
    OUT   - 19
    SCK	  - 18
    CS	  -	 5 
    -----------------
    Other pins:
    RST	  -	 4
    DRDY  -  2 (intr pin)

  CLOCK RATE
	  f_CLKIN = 7.68 MHz
	  τ_CLKIN = 130.2 ns */

/*________________________________________________________________________________________________________
DEFINE REGISTERS 
________________________________________________________________________________________________________*/

//LIBRARY
  #include <SPI.h> //SPI comunication

//BAUD RATE
  #define bdrt_display 2000000

//ADC PINS
  #define RESET_PIN    4
  #define DRDY_PIN     2 
  #define CS_PIN       5 //!!fixed (do not change!)

/*________________________________________________________________________________________________________
DEFINE REGISTERS 
________________________________________________________________________________________________________*/

  #define STATUS  0x00 
  #define MUX     0x01 
  #define ADCON   0x02 
  #define DRATE   0x03 

  // STATUS 
  // ID   - ID   -  ID     -  ID     -  ORDER  -  ACAL   -  BUFEN  -  DRDY 
  #define STATUS_RST  0x01 

  // MUX 
  // PSEL3 - PSEL2 -  PSEL1  -  PSEL0  -  NSEL3  -  NSEL2   - NSEL1   - NSEL0 
  #define MUX_RST     0x01     

  // ADCON
  // 0     - CLK1   -  CLK0   -  SDCS1  -  SDCS0  -  PGA2   -  PGA1   -  PAG0 
  #define ADCON_RST   0x20 
  #define PGA_1       B00100000  //±5V
  #define PGA_2       B00100001  //±2.5V
  #define PGA_4       B00100010  //±1.25V
  #define PGA_8       B00100011  //±0.625V
  #define PGA_16      B00100100  //±0.3125V
  #define PGA_32      B00100101  //±0.15625V    current 0.5µA
  #define PGA_64_2    B00100110  //±0.078125V   current 2µA
  #define PGA_64_10   B00100111  //±0.078125V   current 10µA

  // DRATE
  // DR7  - DR6    -  DR5    -  DR4    -  DR3    -  DR2    -  DR1    -  DR0 
  #define DRATE_RST   0xF0 
  #define DR_100      B10000010 
  #define DR_50       B01100011
[... 262 lines cut ...]
    writeReg(ADCON, PGA_2);
    pga=2;
  }
  else {}

  dynamicVoltBufr = Alot();
  
  writeReg(ADCON, PGA_1); //set pga back to default
  return dynamicVoltBufr;
}

float Alot()
{
  int32_t timesBfr = 100;
  float SigmaVoltBfr = 0.;
  for(int32_t i=1; i<=timesBfr; i++)
  {
    SigmaVoltBfr=SigmaVoltBfr+readVolt();
  }
  SigmaVoltBfr=SigmaVoltBfr/timesBfr;
  return SigmaVoltBfr;
}
`

I've successfully implemented and tested my code and was interested in enhancing the ads1256 library by contributing additional features to make it even more robust. I'm a bit puzzled by your response, and I hope my intention is clear. I believe constructive collaboration can lead to the best possible outcome for the library. I hope you don't take offense; I'm just expressing my eagerness to contribute positively to the project.

CuriousScientist0, 2024-01-31:
First of all, the above code has nothing to do with my library, it is a totally different code. The purpose of my library is to implement all the functions of the ADS1256. No less, no more. Therefore, while your idea might be good, it is not necessary to be included in the library. Those who want this function can implement the logic on top of the library, directly in their Arduino code.

srisankaran, 2024-02-27:
> Do it, if you want, I won't do it for you.

Your comment comes across as unprofessional. I didn't intend to imply that I want additional features for my own purposes; rather, I simply believed it would be beneficial if such a feature existed. As you've indicated, the primary objective of the library is to fully implement all functions of the ADS1256 without compromise. I understand and respect that perspective. However, if this is the tone conveyed through this forum, I question its suitability for constructive dialogue.