CuriousScientist0/ADS1256 #6: Unreliable Register Setting

RogerGdot, 2024-02-06:
Hi,
I have a problem with the writing of the register settings. I have to resend several times in a row the register and the value I want to set, otherwise the the value of the register will not change. Is there a way to change the timing? Everything else works fine

CuriousScientist0, 2024-02-08:
Hi! 
Could you be a bit more detailed? Which MCU? How did you connect the ADS1256 with the MCU? Soldering, breadboard...etc.? I will check it on my side, but so far, I have no issues with it. But yes, maybe it has something to do with the timing, however, I actually manually checked _every_ function with a logic analyzer and the timings seemed OK.

RogerGdot, 2024-02-13:
Hi,

apologies for the delayed response. I needed to ensure and experiment with several combinations. I also noticed that you updated to version 1.1.0 and used that one.

The current situation is as follows: Changing the value of an ADS1256 register requires several command send attempts to store this value in memory. Sometimes it works on the first attempt, but more often it takes the 4th or 5th attempt. I have tried different combinations, including the ADS board from ICTech mentioned in your post, a more expensive one from Waveshare, and another purchased from Amazon. However, all of them show the same problem.

I'm using various microcontrollers, including Arduino, Teensy, and ESP32, and I've tested both breadboard setups and soldered combinations with a Teensy. Interestingly, the only system that works consistently is the Raspberry Pi. It is able to change the register of all ADS1256 boards without any issues, but it is not your library and is written in Python. So it can’t be the ADS boards with the problem.

I have found out that when having the CS pin of the ADS boards on GND, everything works fine, but that’s no satisfying future solution. It definitely has something to do with the CS. I have played with the t6 timing and was sure to have the best GND conditions. Nothing helped. What is so different to the Raspberry and the other MCUs besides Python? This is strange.

Do you have any suggestions or any clue?

Best regards,
Gregor

CuriousScientist0, 2024-02-13:
Hi!

Yes, I pushed a new update because I got some contributions and I also got some time to fix the issues around the PGA which was spotted by another user. 

I still think that the issue is on your side. I have now 10 different microcontrollers and at least 5+3 different ADS1256 boards (+3 is my 3 iterations with the Atmega32U4 MCU) which work just fine. 

How do you instantiate the ADS1256? What do you have in the argument?

closed 2024-02-13 by CuriousScientist0