CuriousScientist0/ADS1256 #10: Reading incorrect values from register (ESP32)

sam-hud, 2024-05-14:
Using the ESP32-S3 and the ADS1256_Example.ino project, the value read from the register after setting it (for example, DRATE) does not match. It appears that the value read from the register is inconsistent or random.

sam-hud, 2024-05-14:
This can be fixed by changing delayMicroseconds in the readRegister() method from 5 to 7.

This value was obtained from the ADS1255-ADS1256 library which defines the read delay as t6 delay = 4*tCLKIN 50*0.13 = 6.5 us

CuriousScientist0, 2024-05-14:
Hi! 
Just to double-check. Have you tested this, or you are just comparing it with someone else's code? I am asking because I tested the code with an ESP32-WROOM-32 and I haven't noticed any issues (and the same applies to the 8 other MCUs). I will take another look at it because I will soon be able to test the code with a WEMOS D1. Thanks!

sam-hud, 2024-05-14:
Hi,This was tested specifically on the Sparkfun Thing Plus ESP32-S3. Though I believe it was also a problem on the Thing Plus ESP32-WROOM (I will have to check this again).Thanks!

arun-krishnan-k, 2024-10-04:
Hi,
I am facing the same issue in reading the register values. I am using Teensy 4.1

arun-krishnan-k, 2024-10-04:
Thanks for the reply.
It is soldered with a custom pcb

RiverWalker92, 2025-06-30:
Also using the Teensy 4.1. Reading ony 0 from the registers using the example code. Any suggestions of what to check?