CuriousScientist0/ADS1256 #21: confirming ADS1256 pinouts for your RP2040 board

jbeale1, 2025-03-27:
Is it correct that the pinout settings in this ADS1256-example.ino file with the comment "RP2040 Waveshare Mini" should work on your custom PCB (in your PCB-Way project) that is using the RP2040 chip?  I had your board design built and stuffed for me by PCB-Way, and the Pico part of the board programs and works just fine as expected. However, there is some connection problem with the onboard ADS1256 chip, in which the Pico is always receiving 0x00 bytes from the ADC chip. I will track this down, I don't expect you to debug my hardware; I just wanted to confirm the existing Github code was expected to work with your existing PCB-way hardware layout. Thanks!

CuriousScientist0, 2025-03-27:
Hi! 
I think you missed watching the video where I show this exact thing: https://www.youtube.com/watch?v=brSo-fb8Lko
Go to 17:53. It will answer your question and solve the issue. ;)

jbeale1, 2025-03-27:
You are quite right. I watched the video a while back, and forgot! :-) But now the board is working. Thank you very much! 

[image]

closed 2025-03-27 by jbeale1

CuriousScientist0, 2025-03-27:
What a nice board! I wonder who designed it... 😄
I am glad to hear that suggested solution worked as expected.