CuriousScientist0/ADS1256 #24: Alternative SPI peripheral?

LsrSal, 2025-05-18:
Trying to use your library with STM32 (F401CC). My hardware constrains forcing me to use second SPI peripheral (IOs PB12-PB15).
Common way to do that is to initialize second instance of the SPI class. But in your library all SPI calls are in CPP file, meaning I cannot do it from the sketch and will have to plow trough library to change SPI peripheral for each call.

Would be nice to have easier way to use alternative SPI hardware peripherals.

Maybe there is a way? Please advise.

Best regards!

CuriousScientist0, 2025-05-18:
Hi! 
Sure, there is a way. You can, for example, modify the constructor to accept a pointer to another SPI bus, then replace the SPI with the pointer to call the methods of the SPI library. With this option, you can select alternative SPI buses as well. But it requires substantial modification.

LsrSal, 2025-05-18:
I guess this is exactly what I did, added right after calling SPI library:
SPIClass SPI_2(PB15, PB14, PB13);

then replaced every single SPI call in the library with SPI_2.

It works!

Unfortunately I can't just redefine first SPI in the sketch. (or do not know how to do it).

I just thought if SPI selection can be simplified as variable or something like polymorphic functions that can be initialized in the sketch - it will be easier for everybody to use, without need to edit library.

Either way, just want to express Thank You for making it!

CuriousScientist0, 2025-05-18:
I am glad that you managed to solve it. 

Yes, it is doable to make it easier to call a different SPI bus from the sketch, but it requires some work. Therefore, I won't close this issue now, and I will modify the library in the near future so it can handle multiple SPIs. Once I implement it, I will write a follow-up here and close the issue. 

Cheers!

CuriousScientist0, 2025-05-29:
@LsrSal I solved the issue. I also tested it specifically with that custom STM32 board, and it works perfectly. Plus, I tested it with ESP32 (SHPI/VSPI), RP2040 (SPI0/SPI1) and other microcontrollers to be sure that it works as intended. 

https://github.com/CuriousScientist0/ADS1256/releases/tag/v1.5.0

closed 2025-05-29 by CuriousScientist0

LsrSal, 2025-05-30:
Really appreciate your work! Thank you!

Tough I did finish my project for now, I will try to recompile it with your update.

I see you have quite a collection of these boards, including your own design. Can I ask you about the performance you are able to get?
What is the noise floor you are getting? Obviously a function of sample rate/sample time/oversampling. At slow sampling with shorted diff inputs I get about 20uV noise out of 5V full scale. I would like to reduce it to 2-3 uV, similar to 34410A (10V scale) that I have  on hand now to compare. I can do statistical filtering, but hopefully there is a straight hardware way to do it.
And the drift/offset. My cold board diff mode before calibration has offset of ~150uV. I can cal it out. But then it is drifting another +30uV as board getting warmer by itself. 34410A also drifts a tiny bit and has dedicated "zero" button. I wonder if this is really unavoidable.