CuriousScientist0/ADS1256 #25: problems with pin settings

mqnc, 2025-09-07:
From the code, it looks like you use 0 to specify that a pin is not used. In some cases that 0 does not get stored in a member:
```c++
	if(RESET_pin !=0) // so we wont copy RESET_pin to _RESET_pin if RESET_pin is 0
	{
	_RESET_pin = RESET_pin;
	pinMode(_RESET_pin, OUTPUT);
	}
```
but then later the potentially undefined _RESET_pin is checked for 0:
```
  //We do a manual chip reset on the ADS1256 - Datasheet Page 27/ RESET
  if(_RESET_pin != 0)
  {
  digitalWrite(_RESET_pin, LOW);
  delay(200);
  digitalWrite(_RESET_pin, HIGH); //RESET is set to high
  delay(1000);
  }

CuriousScientist0, 2025-09-29:
Hi! 
Thanks for the feedback and for pointing out the insufficiencies of the code. I will fix all of these, but allow me a few weeks because I have a little backlog to take care of. I will make sure that the changes won't break the library backwards. Plus, if people can download the update for the library, they are probably skilled enough to change a few numbers in their .ino file. I will make sure that I document the changes carefully. 

I especially like the SPI approach for reading the DRDY. I totally overlooked it. 
For the "no use" pins, I will probably use "-1", I think. The reason I picked 0 is that I started my journey with the ADS1256 by using it with an Arduino Uno. On that microcontroller, you don't want to use pin 0, because it is the RX pin of the serial connection. But you are right. An ESP32 also has pin 0, and it is a legit GPIO port.

CuriousScientist0, 2025-10-22:
I looked into the code. I must say that I won't implement the DRDY part. If you want to use RDATAC, you cannot rely on the status register to get the DRDY status; you must use the DRDY pin. So, there is almost no point in using the status register for getting the DRDY for acquisition. Regarding the not-defined pins, I use "-1"; it seems to work. I applied it to the CS pin as well because one can tie it LOW and use the ADC that way. I will update the documentation in the coming days and upload the new version of the library.